espressif/esp-dev-kits #116: ESP32-S3-USB-OTG HID keyboard example error with bluescreens

lukasboss99, 2024-11-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, 
I have an ESP32-S3_USB-OTG and have tried to receive data via the USB_Host interface by connecting a keyboard to the USB_Host port with the help of the documentation.
For this I used the HID Keyboard Example. Before that I tested the display with the other examples in the ESP IDF, which all worked. However, I could not get the HID Keyboard Example to work. As soon as I opened the Serial Monitor and plugged the USB OTG in and out a few times, I got a blue screen. 
I am also very confused because there is no 5 volts at the USB host port, as 5 volts is standard for USB, and this is needed to use the connected keyboard. 
Does anyone else have these problems or am I using the example incorrectly? Maybe my USB OTG is also broken?

lijunru-hub, 2024-11-18:
Please provide the path to the example code you are using. Are you using the `usb_otg` development board as a USB host to connect to a keyboard? This development board has a USB switch that requires separate GPIO control, and it is configured as a USB device by default. 

You can quickly configure it to USB host mode using the BSP:  
[ESP32-S3-USB-OTG Component](https://components.espressif.com/components/espressif/esp32_s3_usb_otg/versions/1.6.1)

lijunru-hub, 2024-11-18:
use bsp_usb_mode_select_host & bsp_usb_host_power_mode

lukasboss99, 2024-11-18:
I used this code esp-idf/examples/peripherals/usb/host/hid. Yes, I would like to use the USB Otg as a Usb host to connect a keyboard to it. But I have no idea how to configure it as a USB host. I have already tried to manually set the GPIO's in the Arduino IDE to enable the USB host. It was GPIO18 on high