espressif/esp-dev-kits #117: How to accumulate Open AI with ESP32  with esp32-p4-function-ev-board

isthaison, 2025-01-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'd like to talk about OpenAI integration, like the example of ESP32S3-BOX

Horion0415, 2025-01-06:
The latest OpenAI demo for the ESP-BOX is expected to be released soon. However, it is unlikely to be updated on the p4-function-ev-board simultaneously. This part should be relatively easy to port and will likely be updated on other P4 development boards in subsequent releases.

Horion0415, 2025-04-01:
Sorry for the late reply. The AI-related project using WebSocket or WebRTC for P4 is still being optimized and is expected to be released soon. Currently, AI projects such as XiaoZhi AI are based on the S3 platform.

tyeth, 2025-05-05:
Just curious if the offline AI related functionality will also be implemented, or just web-based? (For the P4)

closed 2026-11-15 by isthaison