espressif/esp-dev-kits #118: Please check in dependency lock files for examples

francisduvivier, 2025-01-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

There is no dependency lock files in any of the examples for the p4 devkit. If you try to build one of the examples on a fresh environment, the build is already broken because one of the 0.x dependencies had a breaking change.
So it would be helpful to follow the [guidance from the espressif docs](https://docs.espressif.com/projects/idf-component-manager/en/latest/reference/dependencies_lock.html) and check in the dependency lock files.

Horion0415, 2025-02-06:
Sorry, could you please fully describe the issue you encountered? The dependency lock file is generated during compilation.

francisduvivier, 2025-02-19:
Hi @Horion0415 sorry for the late reply. But so the issue that I encountered was this one: [ESP32-P4 esp_brookesia_phone example broken because of breaking change in esp_ipa (AEGHB-942) #119](https://github.com/espressif/esp-dev-kits/issues/119)

But so I made  separate issue here because it would be quite helpful if those lock files that are generated during compilation are also included in git.
This will really help a lot in the future when examples start getting abandoned, or when somebody wants to go back to an older version of the code. 

I do realise that I linked the wrong documentation link in my description. Apologies for that, I wanted to refer to the [documentation of the dependencies.lock file from Espressif](https://docs.espressif.com/projects/idf-component-manager/en/latest/reference/dependencies_lock.html) which states:

> If your project includes only the [ESP Component Registry Dependencies](https://docs.espressif.com/projects/idf-component-manager/en/latest/reference/manifest_file.html#web-source) or [Git Dependencies](https://docs.espressif.com/projects/idf-component-manager/en/latest/reference/manifest_file.html#git-source), it is recommended to check in the dependencies.lock file to ensure all developers use the same component versions when building locally.