espressif/esp-dev-kits #119: ESP32-P4 esp_brookesia_phone example broken because of breaking change in esp_ipa

francisduvivier, 2025-01-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-dev-1050-gb5ac4fbdf9 but also on v5.4

### Espressif SoC revision.

ESP32-P4 Function_EV_Board V1.4

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-P4 Function_EV_Board V1.4

### Steps to reproduce.

0. Fresh checkout of the repo or remove all git-untracked files in examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone
1. `cd examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone`
2. `idf.py build`

### Build Logs.

```plain
esp-dev-kits/examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp_video/src/esp_video_isp_pipeline.c:752:23: error: too many arguments to function 'esp_ipa_pipeline_create'
  752 |     ESP_GOTO_ON_ERROR(esp_ipa_pipeline_create(config->ipa_nums, config->ipa_names, &isp->ipa_pipeline),
```

### More Information.

The esp32-p4-function-ev-board/examples/esp_brookesia_phone is broken because of a breaking change in the esp_ipa component. 

The esp_ipa component has updated from `0.1.0` 0.2.0 and there are breaking changes which break this project, but the version of esp_ipa is not specified explicitly and there is no dependency lock so now the example is broken.

Note, I also requested lock files to be added for all examples in #118 which could possibly also solve this already, but for this example specifically, I think the esp_ipa 0.1.0 dependency should better also be added to the idf_component.yml specification since it's used directly in this example project.

I added `espressif/esp_ipa: =0.1.0` at the end of my main/idf_component.yml which solves this issue for me.

kylongmu, 2025-01-12:
Same error shows:
D:/Develop/Arduino/ESP32P4/esp-dev-kits-master/examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp_video/src/esp_video_isp_pipeline.c:752:53: error: passing argument 1 of 'esp_ipa_pipeline_create' makes pointer from integer without a cast [-Wint-conversion]
  752 |     ESP_GOTO_ON_ERROR(esp_ipa_pipeline_create(config->ipa_nums, config->ipa_names, &isp->ipa_pipeline),
      |                                               ~~~~~~^~~~~~~~~~

kylongmu, 2025-01-12:
> espressif/esp_ipa:
> version: "0.1.*"

Thanks, you solved this problem.

Horion0415, 2025-01-13:
Sorry, this is a compilation error caused by updates to esp_video and related components. We will update and fix this issue later.

Horion0415, 2025-01-15:
The compilation issue has been fixed. You can test the latest example.

francisduvivier, 2025-01-25:
Example is indeed building again on master.

closed 2025-01-25 by francisduvivier