espressif/esp-dev-kits #120: ESP_BROOKESIA_PHONE fails to build: error: unknown type name 'wifi_tx_rate_config_t'; did you mean 'wifi_init_config_t'

lukilukeskywalker, 2025-01-30:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

ESP32-P4-Function-EV-Board

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-P4-Function-EV-Board

### Steps to reproduce.

When trying to build from scratch the example code, it fails with an error:
`
In file included from /esp-dev-kits/examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp_wifi_remote/include/esp_wifi_remote.h:8,
                 from esp-dev-kits/examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp_wifi_remote/idf_v5.4/esp_wifi_remote_weak.c:7:
esp-dev-kits/examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp_wifi_remote/idf_v5.4/include/esp_wifi_remote_api.h:79:65: error: unknown type name 'wifi_tx_rate_config_t'; did you mean 'wifi_init_config_t'?
   79 | esp_err_t esp_wifi_remote_config_80211_tx(wifi_interface_t ifx, wifi_tx_rate_config_t *config);
      |                                                                 ^~~~~~~~~~~~~~~~~~~~~
      |                                                                 wifi_init_config_t
esp-dev-kits/examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/espressif__esp_wifi_remote/idf_v5.4/esp_wifi_remote_weak.c:369:70: error: unknown type name 'wifi_tx_rate_config_t'; did you mean 'wifi_init_config_t'?
  369 | WEAK esp_err_t esp_wifi_remote_config_80211_tx(wifi_interface_t ifx, wifi_tx_rate_config_t *config)
`
Do I need to enable something?

### Build Logs.

```plain

```

hathorr-hub, 2025-02-03:
I am also facing the same issue .
error: unknown type name 'wifi_tx_rate_config_t'; did you mean 'wifi_init_config_t'?
Can we expect a quick tip for this? We are really stuck for this. 
I am following this blog to create an sample application: https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/docs/esp32_p4_function_ev_board.md
@yogesh.mantri@espressif.com can you please help us for this.

mantriyogesh, 2025-02-04:
1. Have you followed https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/docs/esp32_p4_function_ev_board.md#3-building-host-for-the-p4 ?

2. The `idf_component.yml` and `sdkconfig.defaults.esp32p4` are updated correctly?

3. Also, please ensure to setup **latest idf master** and ensure that you have **updated all git submodules recursively**.
Although these steps mentioned on direct idf.py 

I have not tested with VSCODE IDE although.

mantriyogesh, 2025-02-04:
idf_componen.yml need to be modified to

add:
```
idf.py add-dependency "espressif/esp_wifi_remote"
idf.py add-dependency "espressif/esp_hosted"
```

delete:
```
# ------- Delete or comment this block ---------
### espressif/esp-extconn:
###   version: "~0.1.0"
###   rules:
###     - if: "target in [esp32p4]"
# -----------------------------------
```

hathorr-hub, 2025-02-04:
Hello. 

1. I have taken esp-idf 5.4 

2. this is my idf_component.yml
```
dependencies:
  cmd_system:
    path: ${IDF_PATH}/examples/system/console/advanced/components/cmd_system
  espressif/iperf-cmd:
    version: ~0.1.1
  esp-qa/wifi-cmd:
    version: ~0.1.8
  esp-qa/ping-cmd:
    version: ~1.0.0
  # espressif/esp-extconn:
  #   version: ~0.1.0
  #   rules:
  #   - if: target in [esp32p4]
  espressif/esp_wifi_remote: '*'
  espressif/esp_hosted: '*'
```
3. sdkconfig.defaults.esp32p4 is not generated. I am putting everything inside sdkconfig.default and the content is 
`CONFIG_IDF_TARGET="esp32p4"

CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE=4096

CONFIG_FREERTOS_UNICORE=n
CONFIG_FREERTOS_HZ=1000

CONFIG_ESP_INT_WDT=n
CONFIG_ESP_TASK_WDT_EN=n

CONFIG_LWIP_ETHARP_TRUST_IP_MAC=n
CONFIG_LWIP_IRAM_OPTIMIZATION=y
CONFIG_LWIP_TCPIP_TASK_PRIO=23

CONFIG_IPERF_TRAFFIC_TASK_PRIORITY=23
CONFIG_IPERF_REPORT_TASK_PRIORITY=24
CONFIG_COMPILER_OPTIMIZATION_PERF=y

#### Add Wi-Fi Remote config for better performance:
CONFIG_ESP_WIFI_STATIC_RX_BUFFER_NUM=16
CONFIG_ESP_WIFI_DYNAMIC_RX_BUFFER_NUM=64
CONFIG_ESP_WIFI_DYNAMIC_TX_BUFFER_NUM=64
CONFIG_ESP_WIFI_AMPDU_TX_ENABLED=y
CONFIG_ESP_WIFI_TX_BA_WIN=32
CONFIG_ESP_WIFI_AMPDU_RX_ENABLED=y
CONFIG_ESP_WIFI_RX_BA_WIN=32

CONFIG_LWIP_TCP_SND_BUF_DEFAULT=65534
CONFIG_LWIP_TCP_WND_DEFAULT=65534
CONFIG_LWIP_TCP_RECVMBOX_SIZE=64
CONFIG_LWIP_UDP_RECVMBOX_SIZE=64
CONFIG_LWIP_TCPIP_RECVMBOX_SIZE=64

CONFIG_LWIP_TCP_SACK_OUT=y

CONFIG_ESP_WIFI_ENABLED=y
`
Is there anything I am missing?

mantriyogesh, 2025-02-04:
Can you please delete the the generated sdkconfig file and build/ folder , and try to build again for P4?

Also please attach full textual build log from start, also attach idf_component.yml, sdkconfig.default and generated sdkconfig file

hathorr-hub, 2025-02-04:
Please find the attached details. 

build log: 
[LogForEsp.txt](https://github.com/user-attachments/files/18652335/LogForEsp.txt)

sdkconfig files
[iperf.zip](https://github.com/user-attachments/files/18652340/iperf.zip)

main folder:
[main.zip](https://github.com/user-attachments/files/18652360/main.zip)

SohKamYung-Espressif, 2025-02-04:
@hathorr-hub From the LogForEsp.txt file, I see this (line 23)

```
-- Building ESP-IDF components for target esp32
```
This is the wrong target. Did you do `idf.py set-target esp32p4`?

mantriyogesh, 2025-02-04:
Additionally, I did not understand, why 

```
CONFIG_ESP_WIFI_ENABLED=y
```

Explicitly mentioned in `sdkconfig.defaults`?

Please **remove** this line and 
```sh
rm -rf sdkconfig build/
idf.py set-target esp32p4
idf.py build
```

You can also move it into sdkconfig.defaults.esp32 or other chips, if you need to use single chip solution.
in case of dual chip solution, ideally this variable should not be there.

mantriyogesh, 2025-02-04:
in general, if you stick to instructions in https://github.com/espressif/esp-hosted-mcu/blob/main/docs/esp32_p4_function_ev_board.md, it should just work.

Also, I hope you have c6 pre-flashed (unchanged) and it is hosted slave build.

If it is pre-flashed & hosted, we also recommend you to upgrade the slave firmware using [slave OTA update](https://github.com/espressif/esp-hosted-mcu/blob/main/docs/esp32_p4_function_ev_board.md#51-ota-updates)

lukilukeskywalker, 2025-02-04:
> 1. Have you followed https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/docs/esp32_p4_function_ev_board.md#3-building-host-for-the-p4 ?
>
>     2. The `idf_component.yml` and [...]

I have followed these instructions, have the latest version (fresh instalation of esp-idf 5.4)  and did change the sdkconfig as it was described. I will have another look this evening

andreyvoronko, 2025-02-04:
Hello.
I have the same problem. ESP-IDF 5.4, ESP32P4
In which file is the wifi_interface_t type defined? it is not in any espressif__esp_hosted or project file

andreyvoronko, 2025-02-04:
if you install esp idf v5.4 from vscode plugin, then esp_wifi_types_generic.h file does not contain structure

typedef struct {
    wifi_phy_mode_t phymode;                 /**< Phymode of specified interface */
    wifi_phy_rate_t rate;                    /**< Rate of specified interface */
    bool ersu;                               /**< Using ERSU to send frame, ERSU is a transmission mode related to 802.11 ax.
                                                  ERSU is always used in long distance transmission, and its frame has lower rate compared with SU mode */
    bool dcm;                                /**< Using dcm rate to send frame */
} wifi_tx_rate_config_t;

if you clone esp idf from git, then this structure is there and the project is built.

mantriyogesh, 2025-02-04:
if you clone esp idf from git, then this structure is there and the project is built

So should we assume, this issue is only observed on vscode ide?

lukilukeskywalker, 2025-02-04:
I did look into it because I Did find it really strange.... but @andreyvoronko is right, although not specific to vscode.
The release we can download via `git clone -b v5.4 --recursive https://github.com/espressif/esp-idf.git esp-idf-v5.4` does not contain https://github.com/espressif/esp-idf/blob/release/v5.4/components/esp_wifi/include/esp_wifi_types_generic.h#L1328 the struct (Nor that one or the 2 above it...)

hathorr-hub, 2025-02-04:
Same here I have also clone and check. The structure is not available . Please suggest what next we can do. The error remains same.

hathorr-hub, 2025-02-04:
@lukilukeskywalker can you please let me know the git url to clone as you have already figured the right path.

mantriyogesh, 2025-02-04:
My question is have you tried / tested on latest master instead of 5.4?

andreyvoronko, 2025-02-04:
> So should we assume, this issue is only observed on vscode ide?

I can't answer this question. Maybe the vscode plugin downloads some wrong version? It doesn't clone the repository, it downloads a zip file.

> My question is have you tried / tested on latest master instead of 5.4?

If you enable the master branch, then version 5.5 is enabled, and I couldn't compile with it. Therefore, I downloaded from git b and switched to branch 5.4

hathorr-hub, 2025-02-04:
@mantriyogesh  yes it is working on master. When we can expect it will be available on v5.4+

mantriyogesh, 2025-02-04:
@SohKamYung-Espressif , can we check and do something on hosted and/or wifi remote front?

@hathorr-hub , will get back on this..

SohKamYung-Espressif, 2025-02-05:
This is the commit that adds `wifi_tx_rate_config_t` to `esp_wifi_types_generic.h`

https://github.com/espressif/esp-idf/commit/f890bbe6162f2dc083a683df33e73fb15c358cfb

It is found in the `release/v5.4` branch, and was probably added after the `v5.4` tag.

@hathorr-hub Can you try downgrading wifi-remote to an earlier version? Modify the `idf_component.yml` for wifi-remote to this:

```yaml
  espressif/esp_wifi_remote:
    version: "0.5.4"
```

This earlier version does not include the `esp_wifi_config_80211_tx` API call.

Then,

remove the `managed_components` directory

do

`idf.py fullclean`

Then build the project again.

hathorr-hub, 2025-02-06:
Hello Thanks for the input. Yes I have tested it is working as per the expectation.

mantriyogesh, 2025-02-06:
@hathorr-hub anyway your issue seems to be solved now, it is better to raise different issue next time and cross reference to related bug.. It helps us to focus and avoid cutter.

Now, @lukilukeskywalker , the original auther, do you still face any issues ?

mantriyogesh, 2025-02-06:
As i see, you still face build failure on 5.4 (https://github.com/espressif/esp-dev-kits/issues/120#issuecomment-2634375847) despite of solution from https://github.com/espressif/esp-dev-kits/issues/120#issuecomment-2635562485 ?

If not, we can close this issue?

mantriyogesh, 2025-02-06:
@esp-momo , can you please transfer this issue to https://github.com/espressif/esp-hosted-mcu

lukilukeskywalker, 2025-02-06:
> As i see, you still face build failure on 5.4 ([#120 (comment)](https://github.com/espressif/esp-dev-kits/issues/120#issuecomment-2634375847)) despite of solution from [#120 (comment)](https://github.com/espressif/esp-dev-kits/issues/120#issuecomment-2635562485) ?
> 
> If not, we can close this issue?

I did only pull the esp-idf v5.4 from github and didn't test it yet with the master branch, Haven't had time to do it yet. 
I guess it will work as the only failure I had was that it couldn't find the mentioned structs, so on my side we can close this issue. I will try to try it again this weekend, If I have any other problems I can reopen it or open a new issue in case it fails somewhere else

mantriyogesh, 2025-02-06:
Sure, Thanks for closure. 

For ESP-Hosted-MCU issues, please open at https://github.com/espressif/esp-hosted-mcu/issues, we actively track there for hosted project.

wurei, 2025-02-07:
The commits in the release/v5.4 branch are ahead of those in the tag/5.4. Please use the release/v5.4 branch.

lukilukeskywalker, 2025-02-10:
> release/v5.4

that works!!!

SohKamYung-Espressif, 2025-03-13:
@lukilukeskywalker @hathorr-hub @andreyvoronko If you are still using ESP-IDF v5.4 tag, it should be safe to update wifi-remote to v0.7.x. This version has a fix to support ESP-IDF v5.4 tag release.

In my `idf_component.yml` for esp_brookesia_phone:

```yml
## IDF Component Manager Manifest File
dependencies:
  espressif/esp_wifi_remote:
    version: "~0.7.0"
[...]
```

The esp_brookesia_phone example builds without issue.