espressif/esp-dev-kits #122: esp-dev-kits/examples/esp32-s3-lcd-ev-board/examples  /smart_panel/ fail with ESP-IDF v5.4

klaatu2a, 2025-02-19:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3_LCD_EV_BOARD v1.1

### Power Supply used.

USB

### What is the expected behavior?

Update ESP-IDF to version v5.4.  
Smart_Panel example Compiles, flashes, but fails to run

```
E (2065) wifi:Config authmode threshold is invalid, -1515870811
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x4200d771
--- 0x4200d771: wifi_init_sta at E:/ESPIDF/esp32-s3-lcd-ev-board2/examples/smart_panel/main/app/app_wifi.c:208 (discriminator 1)

file: "./main/app/app_wifi.c" line 208
func: wifi_init_sta
expression: esp_wifi_set_config(WIFI_IF_STA, &wifi_config)

abort() was called at PC 0x40380c47 on core 0
--- 0x40380c47: _esp_error_check_failed at E:/ESPIDF/v5.4/esp-idf/components/esp_system/esp_err.c:49
```

On the readme page for the smart_panel example, it states:

wifi Configuration
    ESP_WIFI_SSID: The name of default Wi-Fi AP.
    ESP_WIFI_PASSWORD: The password of default Wi-Fi AP.

They are no longer in menuconfig

### What is the actual behavior?

The boot screen appears, then it reboots over and over.

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain

```

klaatu2a, 2025-02-20:
I commented out  this line
file: "./main/app/app_wifi.c" line 208

And it worked.  The wifi seems to work too,

GrantGeno, 2025-02-20:
> I commented out this line file: "./main/app/app_wifi.c" line 208
> 
> And it worked. The wifi seems to work too,

If you unplug the board (power it down), then try again, does it work?  Curious how the wifi works without that call, but maybe it's a duplicate that's also done elsewhere...

klaatu2a, 2025-02-20:
> > I commented out this line file: "./main/app/app_wifi.c" line 208
> > And it worked. The wifi seems to work too,
> 
> If you unplug the board (power it down), then try again, does it work? Curious how the wifi works without that call, but maybe it's a duplicate that's also done elsewhere...

Yes,  No issues.    Wifi evenconnected to the access point on its own

klaatu2a, 2025-02-20:
> > I commented out this line file: "./main/app/app_wifi.c" line 208
> > And it worked. The wifi seems to work too,
> 
> If you unplug the board (power it down), then try again, does it work? Curious how the wifi works without that call, but maybe it's a duplicate that's also done elsewhere...

There appears to be no incentive to upgrade to a new version of ESP-IDF.   Its only met with problems.  And there is so much code and so many layers, it crushing to go through it all.   I have a LVGL project, and its a mess.... with v5.4 it does not run.

GrantGeno, 2025-02-20:
Thanks for the heads-up. I see the releases come out in an interesting order.

isamu2025, 2025-03-31:
https://github.com/isamu2025/86box_smart_panel-modify/blob/main/main/app/app_wifi.c