espressif/esp-dev-kits #123: esp-dev-kits esp32-s3-lcd-ev-board boot loops due to wifi config

GrantGeno, 2025-02-19:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1 tested but I see no new changes/improvements

### Espressif SoC revision.

ESP32-S3 mounted on the ESP32-S3-LCD-EV-Board v1.5 and possible others

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-LCD-EV-Board v1.5 

### Power Supply used.

USB

### What is the expected behavior?

The demo should run. The demo does not run. it boot loops. The hardware can be flashed with the pre-compiled binary via 

https://espressif.github.io/esp-launchpad/?flashConfigURL=https://dl.espressif.com/AE/esp-dev-kits/config.toml&app=smart_panel

And that works fine. 

### What is the actual behavior?

The monitor records the assert:

```
E (2332) wifi:Config authmode threshold is invalid, -1515870811
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x4200c9fe
--- 0x4200c9fe: wifi_init_sta at D:/Grant4/ESP32/Examples/esp-dev-kits/examples/esp32-s3-lcd-ev-board/examples/smart_panel/main/app/app_wifi.c:208 (discriminator 1)

file: "./main/app/app_wifi.c" line 208
```
Note: Line number may be slightly off due to my attempted fixes, but I believe the issue is lack of threshold initialization:

```
  wifi_config_t wifi_config;

    sys_param_t *sys_param = settings_get_parameter();
    memcpy(wifi_config.sta.ssid, sys_param->ssid, sizeof(wifi_config.sta.ssid));
    memcpy(wifi_config.sta.password, sys_param->password, sizeof(wifi_config.sta.password));
```
When I supply a value for 

`wifi_config.sta.threshold.authmode = 
`
The demo runs OK past the line

`ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &wifi_config) );`

[... 421 lines cut ...]
I (2130) wifi_init: tcp tx win: 5760
I (2133) wifi_init: tcp rx win: 5760
I (2138) wifi_init: tcp mss: 1440
I (2141) wifi_init: WiFi/LWIP prefer SPIRAM
I (2146) wifi_init: WiFi IRAM OP enabled
I (2151) wifi_init: WiFi RX IRAM OP enabled
E (2336) wifi:Config authmode threshold is invalid, -1515870811
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x4200c9fe
--- 0x4200c9fe: wifi_init_sta at D:/Grant4/ESP32/Examples/esp-dev-kits/examples/esp32-s3-lcd-ev-board/examples/smart_panel/main/app/app_wifi.c:208 (discriminator 1)

file: "./main/app/app_wifi.c" line 208
func: wifi_init_sta
expression: esp_wifi_set_config(WIFI_IF_STA, &wifi_config)

abort() was called at PC 0x40380a13 on core 0
--- 0x40380a13: _esp_error_check_failed at C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_system/esp_err.c:49

Backtrace: 0x40376062:0x3fcc1780 0x40380a1d:0x3fcc17a0 0x40388512:0x3fcc17c0 0x40380a13:0x3fcc1830 0x4200c9fe:0x3fcc1860 0x4200cd4f:0x3fcc1a30
--- 0x40376062: panic_abort at C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_system/panic.c:463
0x40380a1d: esp_system_abort at C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_system/port/esp_system_chip.c:92
0x40388512: abort at C:/Espressif/frameworks/esp-idf-v5.3.1/components/newlib/abort.c:38
0x40380a13: _esp_error_check_failed at C:/Espressif/frameworks/esp-idf-v5.3.1/components/esp_system/esp_err.c:49
0x4200c9fe: wifi_init_sta at D:/Grant4/ESP32/Examples/esp-dev-kits/examples/esp32-s3-lcd-ev-board/examples/smart_panel/main/app/app_wifi.c:208 (discriminator 1)
0x4200cd4f: network_task at D:/Grant4/ESP32/Examples/esp-dev-kits/examples/esp32-s3-lcd-ev-board/examples/smart_panel/main/app/app_wifi.c:220
```

GrantGeno, 2025-02-19:
See also #122. That person is using a newer framework.

isamu2025, 2025-03-31:
itHub Copilot
从错误日志可以看出，WiFi 配置中的认证模式(authmode)设置无效。我们需要正确初始化 WiFi 配置结构体。以下是修复建议：
static void wifi_init_sta(void)
{
    // ...existing code...

    wifi_config_t wifi_config = {
        .sta = {
            .threshold.authmode = ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD,
            .sae_pwe_h2e = WPA3_SAE_PWE_BOTH,
            .pmf_cfg = {
                .capable = true,
                .required = false
            },
        },
    };

    sys_param_t *sys_param = settings_get_parameter();
    memcpy(wifi_config.sta.ssid, sys_param->ssid, sizeof(wifi_config.sta.ssid));
    memcpy(wifi_config.sta.password, sys_param->password, sizeof(wifi_config.sta.password));

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &wifi_config));
    ESP_ERROR_CHECK(esp_wifi_start());
    
    // ...existing code...
}
static void wifi_reconnect_sta(void)
{
    int bits = xEventGroupWaitBits(s_wifi_event_group, WIFI_CONNECTED_BIT, 0, 1, 0);

    wifi_config_t wifi_config = {
        .sta = {
            .threshold.authmode = ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD,
            .sae_pwe_h2e = WPA3_SAE_PWE_BOTH,
            .pmf_cfg = {
                .capable = true,
                .required = false
            },
        },
    };

    sys_param_t *sys_param = settings_get_parameter();
    memcpy(wifi_config.sta.ssid, sys_param->ssid, sizeof(wifi_config.sta.ssid));
    memcpy(wifi_config.sta.password, sys_param->password, sizeof(wifi_config.sta.password));

    // ...existing code...
}

isamu2025, 2025-03-31:
这个错误是因为 ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD 未定义。让我们修复这个问题。

需要在 wifi_reconnect_sta() 和 wifi_init_sta() 函数中修改认证模式的设置。以下是修改建议：

static void wifi_reconnect_sta(void)
{
    int bits = xEventGroupWaitBits(s_wifi_event_group, WIFI_CONNECTED_BIT, 0, 1, 0);

    wifi_config_t wifi_config = {
        .sta = {
            .threshold.authmode = WIFI_AUTH_WPA2_PSK, // 改为具体的认证模式
            .sae_pwe_h2e = WPA3_SAE_PWE_BOTH,
            .pmf_cfg = {
                .capable = true,
                .required = false
            },
        },
    };

    // ...existing code...
}

static void wifi_init_sta(void)
{
    // ...existing code...

    wifi_config_t wifi_config = {
        .sta = {
            .threshold.authmode = WIFI_AUTH_WPA2_PSK, // 改为具体的认证模式
            .sae_pwe_h2e = WPA3_SAE_PWE_BOTH,
            .pmf_cfg = {
                .capable = true,
                .required = false
            },
        },
    };

    // ...existing code...
}

isamu2025, 2025-03-31:
https://github.com/isamu2025/86box_smart_panel-modify/blob/main/main/app/app_wifi.c