espressif/esp-dev-kits #124: esp32p4 ESP_BROOKESIA_PHONE open can`t open the VideoPlayer

Super-DDG, 2025-03-15:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

idf5.4

### Espressif SoC revision.

esp32p4

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32p4_function_ev_borad

### Power Supply used.

USB

### What is the expected behavior?

我今天从仓库里下载了esp32p4的ESP_BROOKESIA_PHONE例子，我在menuconfig中打开了SD卡，重新编译后，我在屏幕上点击Video Player软件时，屏幕变为了浅蓝色，而且串口输出了报错：
```Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4fc1acd8  RA      : 0x481ad882  SP      : 0x4ff386e0  GP      : 0x4ff1b600  
--- 0x4fc1acd8: strncat in ROM
0x481ad882: _svfprintf_r at /builds/idf/crosstool-NG/.build/riscv32-esp-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1521

TP      : 0x4ff38ee0  T0      : 0x4fc19b4c  T1      : 0x4ff387fc  T2      : 0x488611c0  
--- 0x4fc19b4c: __sfputs_r in ROM

S0/FP   : 0x00000000  S1      : 0x00000000  A0      : 0x00000000  A1      : 0xffffffff  
A2      : 0x00000002  A3      : 0x7f7f7f7f  A4      : 0x00000000  A5      : 0x00000004  
A6      : 0x00000001  A7      : 0x00000001  S2      : 0x485f5000  S3      : 0x00000001  
S4      : 0xffffffff  S5      : 0x00000002  S6      : 0x4ff38804  S7      : 0x00000073  
S8      : 0x00000001  S9      : 0x00000000  S10     : 0x00000007  S11     : 0x00000000  
T3      : 0x00020000  T4      : 0x488611b8  T5      : 0x00020001  T6      : 0x488610a8  
MSTATUS : 0x00011880  MTVEC   : 0x4ff00003  MCAUSE  : 0x00000005  MTVAL   : 0x00000000  
--- 0x4ff00003: _vector_table at ??:?

MHARTID : 0x00000000

Stack memory:
4ff386e0: 0x4835c234 0x4ff34a3c 0x4ff38b78 0x481df069 0x00000000 0x00000000 0x00000000 0x00000000
4ff38700: 0x00000000 0x00000001 0x00000008 0x00000007 0x00000000 0x481ddf2c 0x00000000 0x4ff38b60
4ff38720: 0x00000000 0x00000000 0x0001000d 0x00040000 0x00000001 0x00000000 0x00000007 0x4809466c
--- 0x4809466c: lv_txt_get_width at /mnt/d/espidf_project/esp32-p4/esp-dev-kits/examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/lvgl__lvgl/src/misc/lv_txt.c:367

4ff38740: 0x00001f61 0x00000f04 0x00005b41 0x0000000c 0x4ff3875c 0x00000000 0x00000000 0x481e263c
4ff38760: 0x00000001 0x4968c406 0x00000000 0x00000000 0x00000000 0x4835c234 0x0000000c 0x48094792
--- 0x48094792: lv_txt_get_size at /mnt/d/espidf_project/esp32-p4/esp-dev-kits/examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/lvgl__lvgl/src/misc/lv_txt.c:123

4ff38780: 0x00000067 0x00000000 0x00000000 0x00000000 0x00000000 0xffffffff 0x00000000 0x00000000
4ff387a0: 0x00000000 0xffffffff 0x00000000 0x00000003 0x488614d8 0x00000000 0x00000000 0x4ff38c04
[... 78 lines cut ...]
4ff38ac0: 0x000f0000 0x00000013 0x4ff38ba4 0x4809952e 0x4ff38bc4 0x4ff38bc4 0x4ff38bc4 0x480794b6
--- 0x4809952e: lv_dropdown_event at /mnt/d/espidf_project/esp32-p4/esp-dev-kits/examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/lvgl__lvgl/src/widgets/lv_dropdown.c:662
0x480794b6: lv_obj_event_base at /mnt/d/espidf_project/esp32-p4/esp-dev-kits/examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone/managed_components/lvgl__lvgl/src/core/lv_event.c:100

ELF file SHA256: a8d6272d0

CPU halted.
```

### What is the actual behavior?

点击VideoPlayer后，程序出现报错

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain

```

Horion0415, 2025-03-17:
该示例中的视频播放器目前仅支持播放 mjpeg 格式的视频，转换方法请见：https://github.com/espressif/esp-dev-kits/tree/master/examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone#configuration

Super-DDG, 2025-03-27:
> 该示例中的视频播放器目前仅支持播放 mjpeg 格式的视频，转换方法请见：https://github.com/espressif/esp-dev-kits/tree/master/examples/esp32-p4-function-ev-board/examples/esp_brookesia_phone#configuration

我的视频是mjpeg格式，我放在了内存卡下的video目录，当我点击videoPlayer时，程序就报错了，我的内存卡是16G的

Horion0415, 2025-03-29:
把视频放在 SD 卡顶级目录试一下，从 video 目录拿出来