espressif/esp-dev-kits #125: ESP32p4 building capability with wifi is giving error "esp_wifi_remote_init unsupported"

hathorr-hub, 2025-03-21:
Hello Team ESP,

Motivation is to build an application for wifi connectivity on esp32p4 using c6as slave. 
For that I have followed:

1. First I have followed the link : #https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/docs/esp32_p4_function_ev_board.md . And I am able to build the iperf example successfully.
2. Then I want to implement wifi capability in my application using "esp_wifi_remote". For that I have followed ( #https://github.com/espressif/esp-idf/blob/5cbd2a38/examples/protocols/mqtt/tcp/README.md , #https://github.com/espressif/esp-idf/blob/master/examples/wifi/getting_started/station/main/station_example_main.c) . 
3. In config file I have added 
```
`#### Comment below two lines if present:
# CONFIG_ESP_HOST_WIFI_ENABLED=y
# CONFIG_PARTITION_TABLE_SINGLE_APP_LARGE=y

#### Add Wi-Fi Remote config for better performance:
CONFIG_ESP_WIFI_STATIC_RX_BUFFER_NUM=16
CONFIG_ESP_WIFI_DYNAMIC_RX_BUFFER_NUM=64
CONFIG_ESP_WIFI_DYNAMIC_TX_BUFFER_NUM=64
CONFIG_ESP_WIFI_AMPDU_TX_ENABLED=y
CONFIG_ESP_WIFI_TX_BA_WIN=32
CONFIG_ESP_WIFI_AMPDU_RX_ENABLED=y
CONFIG_ESP_WIFI_RX_BA_WIN=32

CONFIG_LWIP_TCP_SND_BUF_DEFAULT=65534
CONFIG_LWIP_TCP_WND_DEFAULT=65534
CONFIG_LWIP_TCP_RECVMBOX_SIZE=64
CONFIG_LWIP_UDP_RECVMBOX_SIZE=64
CONFIG_LWIP_TCPIP_RECVMBOX_SIZE=64

CONFIG_LWIP_TCP_SACK_OUT=y`
```
Also added 
```
CONFIG_ESP_WIFI_REMOTE_LIBRARY_HOSTED=y
CONFIG_IDF_TARGET="esp32p4"

CONFIG_ESP_HOSTED_ENABLED=y
```
 
5. Now when I build and run I can see an error saying : 
`esp_wifi_remote_weak: esp_wifi_remote_init unsupportedW (4436) esp_wifi_remote_weak: esp_wifi_remote_set_ps unsupported
W (4446) esp_wifi_remote_weak: esp_wifi_remote_set_mode unsupported
W (4456) esp_wifi_remote_weak: esp_wifi_remote_start unsupported`

can u please what I am doing wrong?

mantriyogesh, 2025-03-21:
Instead of using reference from lot different places, just use the latest idf `master` or latest branch commit of `release/v5.4`

And use example 'as is' first from https://github.com/espressif/esp-idf/tree/master/examples/wifi/iperf

without changing anything.

mantriyogesh, 2025-03-21:
Let me know if this example works. The incan suggest very small change to make station example work.

hathorr-hub, 2025-03-21:
@mantriyogesh yes we are already following "ESP Chips without Native Wi-Fi" section as you mention. 
but we don't want to use iperf. 
If you can please fill me in with some furthur info who can be causing this error will be a great help. 
`esp_wifi_remote_weak: esp_wifi_remote_init unsupportedW (4436) esp_wifi_remote_weak: esp_wifi_remote_set_ps unsupported W (4446) esp_wifi_remote_weak: esp_wifi_remote_set_mode unsupported W (4456) esp_wifi_remote_weak: esp_wifi_remote_start unsupported`

mantriyogesh, 2025-03-24:
The log you are getting is because the default weak implementation is not overriden provided by the esp-hosted.

While making changes, you have mixed the EPPP link and hosted. Either one is supported. the slave also would need to be of that particular solution. So just flash the 'hosted slave' as document suggest. and clean and re-flash iperf example at host from esp-idf as is.

Once you have iperf example, I will showcase simple changes to get other example to work.

chegewara, 2025-03-24:
Maybe try switching to more recent version?
```
espressif/esp_wifi_remote: '^0.8.0'
```

mantriyogesh, 2025-03-24:
@hathorr-hub  can try as is, esp-idf iperf to get it working first. if it works fine, later they can think of these type of changes.

@chegewara , if you are asking about esp-idf update of the component, the updates to esp-idf would be in general very slower. so we do it periodically.
We have internal change already pending since more than 15 days. but that should not actually matter as long as your APIs are building fine. but you are right, in case only if building fails, this change may be required.

chegewara, 2025-03-24:
Yes, i was asking him to update component to more recent version.
0.5.4 is quite old one.

There is another thing i actually forgot about.
One of my co-worker, not familiar how to use esp-idf, has been trying to build firmware i made and build it failed.
Here is the procedure:
- download zip-ped version of esp-idf
- try to build firmware
When i asked him to `git clone` esp-idf with full procedure, like install, git update submodules etc, then everything was fine.
All i can remember now it was related to `esp_wifi_remote` component and some build errors.

I can try to find more details if required.
BTW in my case it was v0.7.x i think and i esp-idf 5.4 and i had to downgrade esp_wifi_remote, but with git clone any version is working

hathorr-hub, 2025-04-04:
Hello all, the issue has been resolved. As i have used custom CMAKE for building project . I messed up it when I was integrating with p4. Once I fixed it the issue has been resolved. Thanks @mantriyogesh @chegewara

hathorr-hub, 2025-04-04:
Hello Yogesh,
Thanks for helping !!

One more help I need.
Can you please share an example which I can use to read a data file from
the SDcard in ESP32p4.