espressif/esp-dev-kits #126: #38192

gzlwz, 2025-04-12:
E (1783) tca9554: write_direction_reg(129): Write direction reg failed
E (1789) tca9554: reset(146): Write dir reg failed
E (1795) tca9554: esp_io_expander_new_i2c_tca9554(79): Reset failed

gzlwz, 2025-04-12:
esp32-s3-lcd-ev board

gzlwz, 2025-04-12:
E (1783) tca9554: write_direction_reg(129): Write direction reg failed
E (1789) tca9554: reset(146): Write dir reg failed
E (1795) tca9554: esp_io_expander_new_i2c_tca9554(79): Reset failed
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x4200ea08
--- 0x4200ea08: bsp_io_expander_init at D:/downloads/esp-dev-kits-3/examples/esp32-s3-lcd-ev-board/examples/86box_demo/managed_components/espressif__esp32_s3_lcd_ev_board/src/esp32_s3_lcd_ev_board.c:162 (discriminator 1)

file: "./managed_components/espressif__esp32_s3_lcd_ev_board/src/esp32_s3_lcd_ev_board.c" line 162
func: bsp_io_expander_init
expression: esp_io_expander_new_i2c_tca9554((1), ((0x20)), &io_expander)

abort() was called at PC 0x40380e13 on core 0
--- 0x40380e13: _esp_error_check_failed at D:/Espressif50-ide/frameworks/esp-idf-v5.3.1/components/esp_system/esp_err.c:49

Backtrace: 0x40375f36:0x3fca1810 0x40380e1d:0x3fca1830 0x403871ee:0x3fca1850 0x40380e13:0x3fca18c0 0x4200ea08:0x3fca18f0 0x4200eb79:0x3fca1910 0x4200ebf4:0x3fca19b0 0x4200e28b:0x3fca1a10 0x42009fd1:0x3fca1a30 0x420679f3:0x3fca1a60
--- 0x40375f36: panic_abort at D:/Espressif50-ide/frameworks/esp-idf-v5.3.1/components/esp_system/panic.c:463
0x40380e1d: esp_system_abort at D:/Espressif50-ide/frameworks/esp-idf-v5.3.1/components/esp_system/port/esp_system_chip.c:92
0x403871ee: abort at D:/Espressif50-ide/frameworks/esp-idf-v5.3.1/components/newlib/abort.c:38
0x40380e13: _esp_error_check_failed at D:/Espressif50-ide/frameworks/esp-idf-v5.3.1/components/esp_system/esp_err.c:49
0x4200ea08: bsp_io_expander_init at D:/downloads/esp-dev-kits-3/examples/esp32-s3-lcd-ev-board/examples/86box_demo/managed_components/espressif__esp32_s3_lcd_ev_board/src/esp32_s3_lcd_ev_board.c:162 (discriminator 1)
0x4200eb79: bsp_audio_init at D:/downloads/esp-dev-kits-3/examples/esp32-s3-lcd-ev-board/examples/86box_demo/managed_components/espressif__esp32_s3_lcd_ev_board/src/esp32_s3_lcd_ev_board.c:211
0x4200ebf4: bsp_audio_codec_speaker_init at D:/downloads/esp-dev-kits-3/examples/esp32-s3-lcd-ev-board/examples/86box_demo/managed_components/espressif__esp32_s3_lcd_ev_board/src/esp32_s3_lcd_ev_board.c:234
0x4200e28b: bsp_extra_codec_init at D:/downloads/esp-dev-kits-3/examples/esp32-s3-lcd-ev-board/components/bsp_extra/src/bsp_board_extra.c:197
0x42009fd1: app_main at D:/downloads/esp-dev-kits-3/examples/esp32-s3-lcd-ev-board/examples/86box_demo/main/main.c:36
0x420679f3: main_task at D:/Espressif50-ide/frameworks/esp-idf-v5.3.1/components/freertos/app_startup.c:208

Horion0415, 2025-04-13:
Please check the hardware connections and try updating the ESP-IDF version. I was unable to reproduce the issue on both the release/v5.3 and master branches.

gzlwz, 2025-04-14:
hardware: esp32-s3-lcd-ev-board-mb-v1.0 by espressif #28192 ,working well 
idf : 5.3.1
the error is still,below
"E (1783) tca9554: write_direction_reg(129): Write direction reg failed
E (1789) tca9554: reset(146): Write dir reg failed
E (1795) tca9554: esp_io_expander_new_i2c_tca9554(79): Reset failed"